earlephilhower/ESP8266SAM #1: ESP32 PCM5102a Noise

gamecompiler, 2018-03-09:
When running the Code example ESP8266SAM->Examples->Speak theres noise in the background (more foreground):
Id attached a recorded example of it.

When using ESP8266 Audio and i play sth the noise does not appear.

[noise.wav.zip](https://github.com/earlephilhower/ESP8266SAM/files/1795977/noise.wav.zip)

Do i have to configure sth to eliminate the noise in the foreground?

earlephilhower, 2018-03-09:
Your attached ZIP was a 0-byte wav file.  Maybe uploaded before closing in your recording app?

In any case, remember that this is a speech synthesizer from around 1979, running in something like 5K total space, so the noise you're hearing may actually be expected.  You can check the original port author's website out, I think it has an online version to compare it to.

It's not great-sounding, but it is amazingly intelligible given the simple method it's using...

earlephilhower, 2018-05-24:
Please update ESP8266Audio, a bug was found in #2 which may be your problem.

closed 2018-05-24 by earlephilhower