earlephilhower/ESP8266SAM #3: SAM for SAMD21 (Arduino Zero)

deladriere, 2018-04-09:
Hi Phil 
The SamD21 used in the Arduino Zero has an onboard dac (A0)
Any chance to have SAM running also on it?
Many thanks
Cheers,
Jean-Luc

earlephilhower, 2018-04-09:
Howdy,

Didn't know about that board, just looked it up.

As-is, the code should be able to generate the waveforms, but you'll need to write your own AudioOutput class to use the DAC on the board.

I don't think it's a hard project, but if you're really interested then I think a better start would be to jump to the original SAM sources (see the link in my README).

A lot of work was done to make the SAM core Arduino loop() friendly because on the ESP8266 you need to delay() or exit the loop() quickly on each pass or else the watchdog timer will reboot you.  This is needed to allow the WiFi stack to run.

On the Cortex M0 that's not a problem, so you could take the original SAM sources and run the entire speak() call in one pass that takes 10s of seconds to run.

If you do get something running on the SamD21 let me know and I'll add a link in the README.

Thx
-EFP3

closed 2018-04-09 by earlephilhower

deladriere, 2018-04-10:
Thx for taking the time to answer: I agree it would be better to start from the SAM source but that's beyond my skills and I don't know where to start!