earlephilhower/ESP8266SAM #8: Unable to get any output from SAM

debsahu, 2019-03-29:
I have a PCM5102 connected to ESP8266 and all examples from ESP8266Audio work as expected. This means the wiring is right.

I can't get SAM's Say() to output anything. Any help appreciated.

jotadepicas, 2019-03-30:
Hey @debsahu , I started playing with this library a couple of days ago and I it worked just fine. My setup is different as yours though, I'm not using any DAC, I'm playing it in a speaker directly using the "1 transistor" setup proposed in https://github.com/earlephilhower/ESP8266Audio#software-i2s-delta-sigma-dac-ie-playing-music-with-a-single-transistor-and-speaker (and using AudioOutputI2SNoDAC instead).

However, as I see SAM is not coupled to any library in particular, only the examples are, because SAM just outputs to a stream of bytes that in the examples are directed to ESP8266Audio. Maybe you could debug the example to see if the output looks more or less as expected? If you share your code it could be easier to help you troubleshoot.

debsahu, 2019-03-31:
My wiring:
```
      // [PCM5102 DAC > ESP8266/NODEMCU]
      // -------------------------------
      // VCC  > 3.3
      // 3.3V > 3.3
      // GND  > GND
      // FLT  > GND
      // DMP  > GND
      // SCL  > GND
      // BCK  > GPIO15/D8 (I2S CLOCK)
      // DIN  > GPIO3/RX  (I2S DATA)
      // LCK  > GPIO2/D4  (L/R CONTROL)
      // FMT  > GND
      // XMT  > 3.3
```
Sketch:
```
    
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <SD.h>
#include <ESP8266SAM.h>
#include <AudioOutputI2S.h>

AudioOutputI2S *out = NULL;

void setup()
{
  out = new AudioOutputI2S();
  out->begin();
}

void loop()
{
  ESP8266SAM *sam = new ESP8266SAM;
  sam->Say(out, "Can you hear me now?");
  delay(500);
  sam->Say(out, "I can't hear you!");
  delete sam;
}
```