earlephilhower/ESP8266SAM #9: Interested in adding Spanish support.

jotadepicas, 2019-03-30:
Hi, first of all, great work porting this library and on your other library ESP8266Audio as well 😃 

I'm from Argentina, and I'm thinking it would be great to add Spanish support to ESP8266SAM. I _think_ the existing phonemes should work, and the only thing to change would be the mathching from strings to phonemes, so it would implement Spanish pronounciation by selecting different phonemes than the ones for English. From playing with other libraries and frameworks I know this would not render optimal results, probably will sound a little funny, but maybe it could be a good starting point.

Would you provide any guidelines or ideas on how to achieve this? Thanks!

docteurzoidberg, 2024-06-09:
nice. need same for french !

pablomarquez76, 2024-06-09:
> nice. need same for french !

The rules for the language must be program in "reciter.h". You could modify that file to use the rules of French pronunciation.