earlephilhower/ESP8266SAM #11: Other code hanging in the loop()

imayoda, 2019-04-30:
Hi there, and kudos for the excellent porting (and esp8266audio creation too).

I'm using the NoDAC + SAM successfully but now that the program is getting bigger, I noticed that code seems to not be able to loop anymore.
The only solution I've found so far, is to call SAM once per loop and just make him say nothing.
If I do this the other functions start working magically, like the server.handle() (for the incoming http connections to the webserver) and some simple timers;.

Is there something I am missing?
Any suggestion would be really appreciated.
Thank you ! :)

oferzv, 2019-09-07:
add out->stop(); after delete sam; and out->begin(); before ESP8266SAM *sam = new ESP8266SAM;

imayoda, 2019-09-08:
thanks a lot @oferzv , will try asap and report back :)

imayoda, 2020-03-29:
works a treat, thank you again

closed 2020-03-29 by imayoda