earlephilhower/ESP8266SAM #13: How do you use the different voices? Elf, Robot, etc.

thearduinoguy, 2019-11-02:
How do you use the different voices? Elf, Robot, etc.

lowerclasswarfare, 2020-08-05:
I figured it out with a little searching on google. Found it on this guy's project.[https://gist.github.com/toolboc/9a2829b2c3c4a1294889232226fd61f8](url)

You just have to set "sam->SetVoice(sam->SAMVoice::VOICE_SAM)" and replace VOICE_SAM with whatever voice you want.

Here's some sample code I used for two buttons on a breadboard.

`/*`
`* Setting voices and pitch etc, is kind of cryptic. Poor documentation on library creator's github.`
`* To set the voice, pick one: VOICE_SAM, VOICE_ELF, VOICE_ROBOT, VOICE_STUFFY, VOICE_OLDLADY, VOICE_ET`
`* Then in your code you have to set this: sam->SetVoice(sam->SAMVoice::VOICE_SAM) (replace VOICE_SAM with voice of your choosing)`
`*/`

`#include <Arduino.h>`
`#include <ESP8266SAM.h>`
`#include <AudioOutputI2S.h>`

`AudioOutputI2S *out = NULL;`

`const int button1 = 16;`
`const int button2 = 5;`

`int buttonState1 = 0;`
`int buttonState2 = 0;`

`void setup()`
`{`
  `Serial.begin(9600);`
  `pinMode(button1, INPUT);`
  `pinMode(button2, INPUT);`

  `out = new AudioOutputI2S();`
  `out->begin();`
`}`

`void loop() {`
  `buttonState1 = digitalRead(button1);`
  `buttonState2 = digitalRead(button2);`
  `ESP8266SAM *sam = new ESP8266SAM;`

  `if (buttonState1 == HIGH) {`
    `sam->SetVoice(sam->SAMVoice::VOICE_ELF);`
    `sam->Say(out, "Button one has ben pushed");`
    `Serial.println("button1 pushed");`
  `}`
  `if (buttonState2 == HIGH) {`
    `sam->SetVoice(sam->SAMVoice::VOICE_STUFFY);`
    `sam->Say(out, "Button two has ben pushed");`
    `Serial.println("button2 pushed");`
  `}`
  `delete sam;`
`}`

earlephilhower, 2020-08-06:
Fair enough.  The docs are quite bare and you need to look at the header file to find the enuma.

Would you like to make a PR to update the README.md or header file?

closed 2020-08-06 by earlephilhower