earlephilhower/ESP8266SAM #26: no issue but question

docteurzoidberg, 2024-06-09:
Hi. i could not test for now. still waiting for the correct hardware...

But can you tell me if it's possible to use esp8266sam without locking the main program loop ?

Does the main loop hang when the say method is called ?

Goal is to animate something on a screen while tts is running. could this be achieved ?

Thanks by advance for you answer. 
DrZoid

earlephilhower, 2024-06-10:
The call doesn't return until the audio is completed, sorry.

docteurzoidberg, 2024-06-10:
> The call doesn't return until the audio is completed, sorry.

Many thanks for your answer ! 

I guess my goal is now to sync two mcus (over serial/i2c?). one beeing dedicated for SAM ^^

will try this approach